RobTillaart/MCP_DAC #3: Cannot run unless SPI.begin() is called

CraigMason, 2021-05-25:
Hello,

MCP4911
Arduino Nano 33 BLE

Started integrating this library today instead of manually setting up SPI for controlling a 4911. I note that I cannot get the library to work unless I call `SPI.begin()` within my `setup()`. All calls to `analogWrite` hang.

I hacked in a couple of checks to the constructor to see if [SPI.begin()](https://github.com/RobTillaart/MCP_DAC/blob/master/MCP_DAC.cpp#L28) is called, and it appears to be, but regardless, without `SPI.begin()` in my `setup()`, the SPI comms hang.

Tested using [MCP4921_test.ino](https://github.com/RobTillaart/MCP_DAC/blob/master/examples/MCP4921_test/MCP4921_test.ino) by changing [14](https://github.com/RobTillaart/MCP_DAC/blob/master/examples/MCP4921_test/MCP4921_test.ino#L14) to `MCP4911 MCP;` and also removed the performance test.

Not sure if it's more of an upstream issue, but wanted to log it.

RobTillaart, 2021-05-25:
Interesting find. 
Will try to recreate the issue and investigate.
Might take a few days as I might need to order an MCP4911.

RobTillaart, 2021-05-26:
Possible explanation is that the compiler decides **when** to execute the code from the constructor. Most people (including me) assume it is done before **setup()** is called, but that does not need to be true. Theoretically it can be postponed until the first usage of an instance. 
Another cause could be some premature optimization.

In both cases one need to dive into the generated assembly to see what really happens.

Thanks for your quick test, I will merge the branch right away and publish new release.
If there are other issues, please let me know!

RobTillaart, 2021-05-26:
Develop merged into master so I consider this one solved.
Otherwise feel free to reopen the issue.

closed 2021-05-26 by RobTillaart