RobTillaart/MCP_DAC #8: ESP32 support does not work

StefanL38, 2021-07-31:
I tried to compile the MCP4911_test-example but get the error-message
exit status 1
'selectVSPI' was not declared in this scope

I guess this is due to this if-define-condition that does not evaluate to true

/ ESP32 specific
  #if defined(ESP32)
  void     selectHSPI() { _useHSPI = true;  };
  void     selectVSPI() { _useHSPI = false; };
  bool     usesHSPI()   { return _useHSPI;  };
  bool     usesVSPI()   { return !_useHSPI; };
  // to overrule ESP32 default hw pins...
  void     setGPIOpins(uint8_t clk, uint8_t miso, uint8_t mosi, uint8_t select);
  #endif

sighh!
just another one of those libraries that are shared "as is".

So I'm gonna try this MCP4922-library
https://github.com/michd/Arduino-MCP492X

let's see if I can successfully compile with this library

RobTillaart, 2021-07-31:
Hi, 
this version / functionality is merged today - see #5 - after some days of testing. 

I compiled it with the IDE 1.8.13 and board: ESP dev Module  and the ESP32 Wrover module

Can you tell me what board you used?

RobTillaart, 2021-07-31:
https://github.com/michd/Arduino-MCP492X

A quick look at the code shows it does not support VSPI  / HSPI / ESP32

RobTillaart, 2021-07-31:
TODO: need to write a VSPI example

RobTillaart, 2021-07-31:
@StefanL38 

1. Added an example to the master branch HOW to select VSPI for an ESP32.
https://github.com/RobTillaart/MCP_DAC/tree/master/examples/MCP4921_VSPI
If your board is not an ESP32, VSPI is not supported yet,

2. Added a few lines in the readme.md file to make the order of commands to use VSPI more explicit.

StefanL38, 2021-07-31:
Hello Rob,
thank you for answering so quick.
I made some progress. The error of not compiling was caused by my call of selectVSPI()
just "selectVSPI()"
without a 
MCP.selectVSPI();
now the MCP4911_test-code does compile but I don't see any signal change on any pin of my ESP32
it is a JOY-IT ESP32 DEV-module
I will test your example for the ESP32.
I will report if I have succcess or what questions I still have....
best regards Stefan

StefanL38, 2021-07-31:
Yes I do have questions:
your example uses
`  MCP.setGPIOpins(20,21,22,10);   // needs to be called after begin()
`
But it is completely unclear to me which ESP32-IO-pin number must be connected to what on the MCP4922
best regards Stefan

RobTillaart, 2021-07-31:
> it is a JOY-IT ESP32 DEV-module

It is not in my list of boards, which one  do you use to compile?

StefanL38, 2021-07-31:
ESP32 Dev module. Never had any issue with this board selection
I'm using this pinout picture
[image]
Your call to 
`  //              clk, miso, mosi, select
  MCP.setGPIOpins(20,21,22,10);   // needs to be called after begin()
`
makes no sense with this pinout GPIO20 is not in there
best regards Stefan

RobTillaart, 2021-07-31:
> MCP.setGPIOpins(20, 21, 22, 10);   

The order of the parameters  
- CLOCK,   
- MISO (not used), 
- MOSI, 
- SELECT

The numbers I used are just random chosen (your setup might use other ones).
If you remove this line ```MCP.setGPIOpins(20, 21, 22, 10);```  the VSPI default pins will be used 

See table https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux

RobTillaart, 2021-07-31:
> makes no sense with this pinout GPIO20 is not in there

correct, GPIO10 is also not in the picture, so you must adjust the sketch to the pins you use.

StefanL38, 2021-07-31:
I have a big request can you post an example where real accessible IO-pins are used (not the ones connected to the flash)
and add some kind of documentation which ESP32-IO-pin must be connected to what IO-pin of the MCP4922

The abbreviations all differ and that is the reason why it is unclear to me
[image]

best regards Stefan

RobTillaart, 2021-07-31:
| MCP4922  | ESP32  VSPI   |
|:--------:|:-------------:|
|  CS      |  SELECT = 15  |
|  SCK     |  SCLK   = 14  |
|  SDI     |  MOSI   = 13  |
|

StefanL38, 2021-07-31:
Hi Rob,
thank you very much. I will re-wire my setup and test...

RobTillaart, 2021-07-31:
You're welcome, with a bit of luck and persistence it will work :)

The example has been updated to a sort of minimum which uses the default pins of VSPI.

RobTillaart, 2021-07-31:
Note: I am developing a minor update in the develop branch. 
should not interfere with your issue except that I might be offline ...

StefanL38, 2021-07-31:
I have added this code to loop() because I wanted to measure the voltage of the DAC.
`void loop() {
  MCP.fastWriteA(1024);
  MCP.fastWriteB(1024);
  //Serial.println("MCP.fastWriteA(1024)");
  //delay(3000);
  MCP.fastWriteA(2048);
  MCP.fastWriteB(2048);
  //Serial.println("MCP.fastWriteA(2048)");
  //delay(3000);
}
`
But it always shows 0,00V on my digital multimeter

I commented out the serial print and delays to have a continious stream of command send to the MCP4922
because I want to check if there are any signals on the IO-pins.

If I'm probing the Chip-selectpin
`  MCP.begin(15);
`
I see a 400-500 nanoseconds long high-pulse that repeats every 10 microseconds
I have no experience with SPI at all but this looks strange to me.

The datasheet of the MCP4902/4912/4922 says
3.2 Chip Select (CS)
CS is the Chip Select input, which requires an 
**active low** signal to enable serial clock and data functions.

If I connect the second channel of my 100 MHz digital storage-oscilloscope to MOSI (GPIO13) or clock  (GPIO 14) 
I see **no** pulses at all!?!?!???

So I checked all IO-pins with my DSO but I did not see any signal on any IO-pin.

Next thing I checked was:  if the IO-pins work at all. So I uploaded a small democode that does PWM on the IO-pins

`int freq = 5000;
int ledChannel = 0;
int resolution = 8;

const byte IO_Pin = 13;
void setup() {
  ledcSetup(ledChannel, freq, resolution);
  ledcAttachPin(IO_Pin, ledChannel); 
}
 
void loop() {
  for (int dutyCycle = 0; dutyCycle <= 255; dutyCycle++) {
    ledcWrite(ledChannel, dutyCycle);
    delay(7);
  } 
  for (int dutyCycle = 255; dutyCycle >= 0; dutyCycle--) {
    ledcWrite(ledChannel, dutyCycle);
    delay(7);
  }
}`

My DSO clearly showed PWM based on **ledc** works. The duty-cycle goes up and down like it should

Did you do a **real** hardware-test with your library? 
I mean connecting a **real** MCP4921/MCP4922 to an ESP32 
connecting **all** the wires including SHDN, VrefA, VrefB and **really** measured with a DMM or DSO if the right voltage
is **really created** on the VoutA / VOutB-pins of the MCP49x"?

best regards Stefan

RobTillaart, 2021-07-31:
(put better tags around the code in your post)

> Did you do a real hardware-test with your library?
No, as you can read in issue #5 I did not connect any MCP to my ESP32 as I have no MCP available, however Hank did the testing and he had it working in the end. Not immediately, it took some iterations.

Coming week - I hope Wednesday - I will have and time and the hardware needed - so it is definitely on my list to verify it is working by myself.

Previous versions of the library were tested with UNO and an ESP32 (used the HSPI if I recall correctly)

Do you have pull up resistors on the MOSI and CLOCK line? 
As the library is default set to 16MHz this could be a problem.  try setting the MCP.setSpeed(1000000);

RobTillaart, 2021-07-31:
I think the pin table got switched  VSPI - HSPI columns , sorry

| MCP4922  | ESP32  VSPI   | ESP32  HSPI   |
|:--------:|:-------------:|:-------------:|
|  CS      |  SELECT = 5  |  SELECT = 15   | 
|  SCK     |  SCLK   = 18 |  SCLK   = 14  | 
|  SDI     |  MOSI   = 23 |  MOSI   = 13  | 

I measured and when selecting VSPI - and setSPIspeed(1000000) - I got a clock on pin 18

RobTillaart, 2021-08-01:
merged the 0.1.3 preparations in master
- added **increment(channel)**
- added **decrement(channel)**
- updated **readme.md** to reflect correct SPI pins
- updated **VSPI** example with some comments.

RobTillaart, 2021-08-01:
### TEST 1 

no MCP connected,
when **selectVSPI()**  I get MOSI signals on pin 23, and CLOCK on pin 18
When **selectHSPI()** I get MOSI signals on pin 13 and CLOCK on pin 14

so SPI peripheral selection works and data is send

RobTillaart, 2021-08-01:
merged the 0.1.4 in master
- fix **setGPIOpins()** to overrule default HSPI / VSPI pins

HankLloydRight, 2021-08-03:
> Did you do a **real** hardware-test with your library?
> I mean connecting a **real** MCP4921/MCP4922 to an ESP32
> connecting **all** the wires including SHDN, VrefA, VrefB and **really** measured with a DMM or DSO if the right voltage
> is **really created** on the VoutA / VOutB-pins of the MCP49x"?

RobTillaart, 2021-08-04:
[image]

 ESP32 via VSPI to MCP4922
 external power of 5V
 Sinus 1000 Hz 

MCP4921_wave_generator.ino with these lines added to use the VSPI 
```cpp
  MCP.selectVSPI();     // needs to be called before begin()
                        // uses default HSPI SCLK=14, MISO=12, MOSI=13, SELECT=15
                        // uses default VSPI SCLK=18, MISO=19, MOSI=23, SELECT=5
  MCP.begin(15);        // 5 for VSPI and 15 for HSPI
  ```
  
  works

RobTillaart, 2021-08-05:
@HankLloydRight 

> edit: I'm working on a 6 channel DAC:

You might add a LED to the CS lines to see activity ?

HankLloydRight, 2021-08-05:
Yup! Already in the next layout, this was just a V0 test board.

HankLloydRight, 2021-08-06:
> Yup! Already in the next layout, this was just a V0 test board.

It occurred to me though, at 30kHz, wouldn't the CS line LEDs just seem to be "on" all the time?

RobTillaart, 2021-08-06:
> > Yup! Already in the next layout, this was just a V0 test board.
> 
> It occurred to me though, at 30kHz, wouldn't the CS line LEDs just seem to be "on" all the time?

Yes and that is exactly what you want to see. It informs you that all 3 your channels (x2) get their data in a very high pace.
I always add several LEDs. That allows my customers to send a picture of the board so I have a first order "diagnostics" what works and what not. A LED for 3.3, 5 and/or 12V etc is useful to diagnose a broken power supply or fuse "poofed".

HankLloydRight, 2021-08-06:
I have LEDs on the 3.3v  output of the voltage regulator as well as LEDS on the +5v and -5V output of the bipolar power supply  (not in the photo above).  BTW, how do I tap the CS lines with an LED without affecting the signal? I also have RGB analog output (0-5v) which I'd like to add an LED, but again, don't want to affect the signal.  Thanks.

(I only barely know what I'm doing, I'm not a trained EE :) )

RobTillaart, 2021-08-06:
An LED will not take much current.
Assume you have 5V supply
Assume Led may take 10mA at max 
Assume led needs 2V 

formula: R = (voltage supply - voltage LED) / (desired led current)  

R = (5 - 2) / 0.01 = 300 ohm 

If you want less luminosity you take 600 ohm (~50%) - or a standard value of 470 ohm (~60%).
For diagnostics 60% is often bright enough. Note the leds are already dims as they are not 100% on.

More detailed see e.g. https://www.circuitspecialists.com/blog/how-to-determine-resistor-value-for-led-lighting/

HankLloydRight, 2021-08-06:
Awesome, thank you!

RobTillaart, 2021-08-06:
@StefanL38 
Did you manage to get it working?

RobTillaart, 2021-08-15:
@StefanL38
If the issue persists feel free to reopen the issue.

closed 2021-08-15 by RobTillaart