RobTillaart/MCP_DAC #11: Method to double the DAC write speed in swSPI_transfer()

HankLloydRight, 2021-09-17:
Hello Rob,

 I'm still working on my 6 channel DAC with MCP4922.  I wanted to eek out every possible cpu cycle to maximize the speed of the library.  I was never able to get HW SPI to run faster, because the penalty of starting and ending a transaction was greater than bit-banging on the SW SPI bus.  But I found another way.   I was looking to optimize the core fastWriteA()/fastWriteB(), transfer() and swSPI_transfer() functions.  I dug deeper into the Arduino digitalWrite() function, and it has A LOT of overhead.  I read about using direct I/O registers to toggle the pins instead of digitalWrite().  
 So I went ahead and added some new functions that write directly to the IO registers, and the speed of the library more than doubled.  Here's how I did it:

The first step is to set the registers for writing in setup(). You first initialize the MCP_DAC normally. Then to setup the registers, you use the BIT# macros to do this. In this case, I'm writing to pins 23 (MOSI),5 (CS1),16 (CS2) ,27 (CS3),18 (CLK), and 17 (LATCH). You just use this function:

` REG_WRITE ( GPIO_ENABLE_REG, BIT23+BIT5+BIT16+BIT27+BIT18+BIT17 ) ;// MOSI,XY,RG,BS,CLK,latch `

I then setup to fastWriteA_direct and fastWriteB_direct functions:

```
void MCP_DAC::fastWriteA_direct(uint16_t v1) {
    uint32_t sel=(1 << _select );    
    REG_WRITE ( GPIO_OUT_W1TC_REG, sel);
    swSPI_transfer_directio((uint8_t)((0x3000 | v1) >> 8));
    swSPI_transfer_directio((uint8_t)(v1 & 0xFF));
    REG_WRITE ( GPIO_OUT_W1TS_REG, sel);
}

void MCP_DAC::fastWriteB_direct(uint16_t v2) {
    uint32_t sel=(1 << _select );    
    REG_WRITE ( GPIO_OUT_W1TC_REG, sel);
    swSPI_transfer_directio((uint8_t)((0xB000 | v2) >> 8));
    swSPI_transfer_directio((uint8_t)(v2 & 0xFF));
    REG_WRITE ( GPIO_OUT_W1TS_REG, sel);
}
```

The first line "sel=(1 << _select );  " basically takes 2 and raises it to power of the CS Pin# (2^PIN#) -- this is a shorthand for calculating the BIT# macro values used in the setup() step above.  The second line toggles the IO register LOW (W1TC='Clear').  I call the swSPI_transfer_directio() function to send the data, then toggle the CS register pin high (W1TS='Set').

The transfer function is this:
```
void MCP_DAC::swSPI_transfer_directio(uint8_t val)
{
  uint32_t clk = 1 << _clock;
  uint32_t dao = 1 << _dataOut ;
  for (uint8_t mask = 0x80; mask; mask >>= 1)  {
    if (val & mask) REG_WRITE ( GPIO_OUT_W1TS_REG, dao ) ; 
    else REG_WRITE ( GPIO_OUT_W1TC_REG, dao ) ;
    REG_WRITE ( GPIO_OUT_W1TS_REG, clk);
    REG_WRITE ( GPIO_OUT_W1TC_REG, clk);
  }
}
```

The first two lines do the same trick to calculate the register BIT# for the data pin (MOSI) and clock pins. Then each bit is set HIGH(Set) or LOW(Clear) in turn.  The last two statements just toggle the CLK pin HIGH then LOW.

I'm using the LATCH PIN right now, but that function is updated in exactly the same way:

```
void MCP_DAC::triggerLatch_direct() {
  if (_latchPin != 255)  {
    uint32_t ltch = 1 << _latchPin;
    REG_WRITE ( GPIO_OUT_W1TS_REG, ltch);  
    delayMicroseconds(1);     // 100 ns - Page 7
    REG_WRITE ( GPIO_OUT_W1TC_REG, ltch);
  }
}
```

What's nice is these new functions can live side-by-side with the existing ones, and the user can call whichever ones they want.

The only disclaimer is that this is all done on Platform.io using the Arduino framework.  I do not know how or if this will work on Arduino IDE, but there has to be a way.  

I can now push points to my DACs in upwards of 100kHz. 

-Hank

RobTillaart, 2021-09-18:
Register optimization is a known way to squeeze more speed from a system but one looses portability. That said the results are impressive.
Can  you put the performance data in a table to compare them?

I will check if I can replicate .

BTW did you change pullup resistors for pulse squareness quality?

RobTillaart, 2021-09-18:
A possible way to incorporate this version is to make a derived ESP specific class 

The base class is **MCP_DAC**
From that derive an ESP specific base class **MCP_DAC_ESP** with your optimizations.
From that derive MCP4922_ESP and all others.

RobTillaart, 2021-09-18:
```cpp
void MCP_DAC::swSPI_transfer_directio(uint8_t val)
{
  uint32_t clk = 1 << _clock;
  uint32_t dao = 1 << _dataOut ;
  for (uint8_t mask = 0x80; mask; mask >>= 1)  {
    if (val & mask) REG_WRITE ( GPIO_OUT_W1TS_REG, dao ) ; 
    else REG_WRITE ( GPIO_OUT_W1TC_REG, dao ) ;
    REG_WRITE ( GPIO_OUT_W1TS_REG, clk);
    REG_WRITE ( GPIO_OUT_W1TC_REG, clk);
  }
}
```

The two lines (dao and clk) are executed for every byte transferred.
If you add 2 more variables that are set as soon as the clock and MOSI are set 
```cpp
__clock = (1 << _clock);
__dataout = (1 << _dataout);
```

it would become
```cpp
void MCP_DAC::swSPI_transfer_directio(uint8_t val)
{
  uint32_t clk = __clock;
  uint32_t dao = __dataOut ;
  for (uint8_t mask = 0x80; mask; mask >>= 1)  {
    if (val & mask) REG_WRITE ( GPIO_OUT_W1TS_REG, dao ) ; 
    else REG_WRITE ( GPIO_OUT_W1TC_REG, dao ) ;
    REG_WRITE ( GPIO_OUT_W1TS_REG, clk);
    REG_WRITE ( GPIO_OUT_W1TC_REG, clk);
  }
}
```
No more shift => saving a few clocks per byte transferred - I expect it will hardly noticeable

Another thought might be the datatype of **dao** and **clk**
making it uint16_t or uint8_t might save a few clocks too - depends on how **REG_WRITE()** is implemented

RobTillaart, 2021-09-18:
### wild idea section

(not for the faint of heart)
There is another possible way to optimize (preferably) 8 SPI based DAC's in parallel.
Some processors have byte based IO registers which can write 8 lines at once.

By loading the register with 8 MOSI bits one can clock them in parallal
- selection line must be a shared line
- clock must be a shared line.

You loose some performance setting up the IO byte, but you win some in the clocking / selection.
The extra you get is that the MCP's change their values in perfect sync.

disclaimer: never tried with MCP's

opinion?

HankLloydRight, 2021-09-18:
Thanks for the replies!  Let's see where to start?
>Register optimization is a known way to squeeze more speed from a system but one looses portability

Yes, I realized this right away, but I thought it was such a great improvement that I just wanted to share it with other people. 

> Can you put the performance data in a table to compare them?

I can try to later this week.  It's really just a "before" and "after".  With digitalWrite() I was pushing about 150k DAC updates per second.  An update being one channel/one DAC, I have three DACs and 5 channels making up a "point", so that was 30k "points" per second.  Then after changing to using registers, I was getting about 60k points per second (or 300k DAC updates per second).  I'm now timing those updates, and to update 5 channels takes about 15µs.  

>BTW did you change pullup resistors for pulse squareness quality?

I did not change the hardware at all.  I do not have any pull-up resistors on the DACs.  Should I?  Pulse squareness quality sounds like something I should have.  :)

> (1) A possible way to incorporate this version is to make a derived ESP specific class
>(2) Alternative is to embed the code in the base class guarded with enough defines to automatically select the optimized version.
>(3) Third option is to implement it as a standalone ESP class. (small risk that the interfaces get out of sync)

I could do one or all of these, I'm just not sure the best way to encapsulate or make the first initialization step 'generic' in the scope of using a library:

`REG_WRITE ( GPIO_ENABLE_REG, BIT23+BIT5+BIT16+BIT27+BIT18+BIT17 ) ;`

Perhaps I'll fork your wonderful MCP_DAC repo and try to create an ESP specific one.  I'd still need to figure out how to get it to compile on Arduino IDE though. 

>If you add 2 more variables that are set as soon as the clock and MOSI are set
>No more shift => saving a few clocks per byte transferred - I expect it will hardly noticeable. (~ 0.5% ?)

I'll definitely try this and let you know.  Great suggestion. Thanks.

>There is another possible way to optimize (preferably) 8 SPI based DAC's in parallel.

Yeah, a friend of mine also suggested the same idea. He simply said "have you thought about just swapping the CS and DATA lines and updating all the DACs at the same time."  I had to think about it for a second, but yeah, that sounds like it just might work.  The problem is all my hardware is already made into PCBs.  I'd have to go back to breadboarding (and A LOT of software changes) to test this out.   But it is a very intriguing idea for sure.  If it currently takes 15µs to update 5 channels per cycle, I could in theory get that down to 3-4µs per cycle or a 4-5x increase in throughput.  WOW.

RobTillaart, 2021-09-19:
> I could do one or all of these, I'm just not sure the best way to encapsulate or make the first initialization step 'generic' in the scope of using a library:
```cpp
REG_WRITE ( GPIO_ENABLE_REG, BIT23+BIT5+BIT16+BIT27+BIT18+BIT17 ) ;
```

should be in the **constructor() / begin() / setLatchPin()** e.g 
```cpp
REG_WRITE ( GPIO_ENABLE_REG,  (1 << LATCH) ) ;
```
probably keep things like setting **pinMode()**

Just make an ESP base class derived from  MCP_ADC and start to replace generic with specific code.

HankLloydRight, 2021-09-20:
> ### wild idea section
> (not for the faint of heart)
> There is another possible way to optimize (preferably) 8 SPI based DAC's in parallel.
> Some processors have byte based IO registers which can write [...]

Ok, so I tried this yesterday and today.  And it works!  It took some time to get the timings right, but it works. I have all three DACs on the same CS line, and separate SDI/MOSI lines.  I still had to write to the ESP32 pin registers one bit at a time (same as your existing swSPI_transfer() method).  The end result is that it's about 25% faster.  Under the standard SPI method, the timing to send one point (5 bytes) to the three DAC was 15µs.   With this CS/MOSI swap, it was about 10µs (there's still some overhead, so it's not actually 50% faster).   But this didn't resolve some sample jitter that I'm seeing on the analog outputs I was hoping it would solve.    So I reverted back to the standard SPI pins/code.  It was a fun experiment, for sure. 

>Another thought might be the datatype of dao and clk
>making it uint16_t or uint8_t might save a few clocks too - depends on how REG_WRITE() is implemented

When bit shifting the registers like this, the max value for those could be 2^31=2147483648, so it has to use uint32_t.

While doing this experiment, I realized that 80% of your library code handles the hardware SPI side of things, and very little of the code is for software SPI.  I really could reduce the library to just a few functions if it were only to handle software SPI/bitbanging.  I might just do this and remove all the hardware SPI code into a more streamlined library just for using the registers on ESP32. 

Also, I tried several ways to get the hardware SPI(VSPI) to work, and every time/every way it was significantly slower. Not just your library, but even using native ESP32-idf SPI functions.   I'm sure I'm missing something, but the software SPI interface works great. (I am using the proper SPI3/VSPI pins, so it's not the GPIO mux).

RobTillaart, 2021-09-20:
Good idea to strip things to a software-only library, although I expect the compiler to keep most unused code out of it. 
Still it would be max optimized if stripped. 

> I realized that 80% of your library code handles the hardware SPI side of things, ...

Yes, that is true, as people expect it to be the fastest, but when going as low level as you did the overhead of the (hw) SPI library becomes relative large. (as you noticed)

----

Nice result for the three parallel devices. 

> But this didn't resolve some sample jitter that I'm seeing on the analog outputs I was hoping it would solve. 

Can you elaborate on this jitter, how much / often do you see it?  
Only elaborate if you have time, I think it is interesting as this solution is quite "groundbreaking"

RobTillaart, 2021-10-10:
@HankLloydRight 
Are there still actions expected on this issue or can it be closed?

HankLloydRight, 2021-10-10:
None at all!  I tried to get a video of the sample jitter you asked for, but I need to find a better camera. Everything is working great now, so you can close this issue.  Thank you again for all your help.