RobTillaart/MCP_DAC #13: MCP4921_wave_generator

AGenews, 2021-11-09:
Just came here to let you know the MCP4921_wave_generator does currently no compile for a UNO, though giving me the follwoing error message:

`
/tmp/arduino_modified_sketch_430116/MCP4921_wave_generator.ino: In function 'void setup()':
MCP4921_wave_generator:25:7: error: 'class MCP4921' has no member named 'selectVSPI'; did you mean '_select'?
   MCP.selectVSPI();     // needs to be called before begin()
       ^~~~~~~~~~
       _select
exit status 1
'class MCP4921' has no member named 'selectVSPI'; did you mean '_select'?`

Thanks a ot for all you hard work and the great MCPxx libraries!
Best,
Andreas

RobTillaart, 2021-11-09:
Yes I know I'm working on extending the github automatic build to include ESP32 devices for all of my libraries. 
In this process I update some libraries, fix documentation, and sometimes change examples to do ESP32 specific stuff.
That's why I adapted this sketch to use specifically the VSPI interface. 

proposed solution:
I will clone the sketch to have an AVR/UNO version available and put ESP32 in the name of the specific one.

RobTillaart, 2021-11-09:
Can you confirm this sketch works for UNO (I have no setup nearby)

```cpp
//
//    FILE: MCP4921_wave_generator.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: demo function generators
//    DATE: 2021-02-03
//     URL: https://github.com/RobTillaart/FunctionGenerator
//
//  depending on the platform, the range of "smooth" sinus is limited.
//  other signals are less difficult so have a slightly larger range.
//
//  PLATFORM     SINUS    SQUARE  SAWTOOTH  TRIANGLE
//  UNO          -100 Hz
//  ESP32        -200 Hz  -1000   -250      -100
//

#include "MCP_DAC.h"

uint16_t     freq = 10;
uint32_t   period = 0;
uint32_t   halvePeriod = 0;

// q = square
// s = sinus
// w = sawtooth
// t = triangle
// r = random
char mode = 'q';

MCP4921 MCP;
uint16_t count;
uint32_t lastTime = 0;

// LOOKUP TABLE SINE
uint16_t sine[361];

void setup()
{
  Serial.begin(115200);

  // fill table with sinus values for fast lookup
  for (int i = 0; i < 361; i++)
  {
    sine[i] = 2047 + round(2047 * sin(i * PI / 180));
  }

  MCP.begin(7);  // select pin = 7
  MCP.fastWriteA(0);

  period = 1e6 / freq;
  halvePeriod = period / 2;

  while (1)
  {
    // Serial.println(analogRead(A0));   // read output back via A0.
    yield();
    uint32_t now = micros();

    count++;

    if (now - lastTime > 100000)
    {
      lastTime = now;
      // Serial.println(count); // show # updates per 0.1 second
      count = 0;
      if (Serial.available())
      {
        int c = Serial.read();
        switch (c)
        {
          case '+':
            freq++;
            break;
          case '-':
            freq--;
            break;
          case '*':
            freq *= 10;
            break;
          case '/':
            freq /= 10;
            break;
          case '0' ... '9':
            freq *= 10;
            freq += (c - '0');
            break;
          case 'c':
            freq = 0;
            break;
          case 'A':
            break;
          case 'a':
            break;
          case 'q':
          case 's':
          case 'w':
          case 't':
          case 'r':
          case 'z':
          case 'm':
          case 'h':
            mode = c;
            break;
          default:
            break;
        }
        period = 1e6 / freq;
        halvePeriod = period / 2;
        Serial.print(freq);
        //        Serial.print('\t');
        //        Serial.print(period);
        //        Serial.print('\t');
        //        Serial.print(halvePeriod);
        Serial.println();
      }
    }

    uint32_t t = now % period;

    switch (mode)
    {
      case 'q':
        if (t < halvePeriod ) MCP.fastWriteA(4095);
        else MCP.fastWriteA(0);
        break;
      case 'w':
        MCP.fastWriteA(t * 4095 / period );
        break;
      case 't':
        if (t < halvePeriod) MCP.fastWriteA(t * 4095 / halvePeriod);
        else MCP.fastWriteA( (period - t) * 4095 / halvePeriod );
        break;
      case 'r':
        MCP.fastWriteA(random(4096));
        break;
      case 'z':  // zero
        MCP.fastWriteA(0);
        break;
      case 'h':  // high
        MCP.fastWriteA(4095);
        break;
      case 'm':  // mid
        MCP.fastWriteA(2047);
        break;
      default:
      case 's':
        // reference
        // float f = ((PI * 2) * t)/period;
        // MCP.setValue(2047 + 2047 * sin(f));
        //
        int idx = (360 * t) / period;
        MCP.fastWriteA(sine[idx]);  // lookuptable
        break;
    }
  }
}

void loop()
{
}

// -- END OF FILE --

```

RobTillaart, 2021-11-09:
Created PR - https://github.com/RobTillaart/MCP_DAC/pull/14

AGenews, 2021-11-09:
Dear Rob,
for my setup I just changed MCP.begin(7); to MCP.begin(10); and it worked like a charm!
Thank you for the fix!
All the Best,
Andreas

RobTillaart, 2021-11-09:
OK< pin 10 it will be  😀

commit ec79232d (2021-11-09): fix #13 - add examples
commit 02020278 (2021-11-09): fix #13 - add examples (#14)

* fix #13 - add examples
closed 2021-11-09 by RobTillaart