RobTillaart/MCP_DAC #21: triggerLatch() is inverted

nelisbrink, 2023-03-23:
I am working with the MCP4822 and according to the datasheet the LDAC is a NOT pin. So, quoting from the datasheet: 

"By bringing down the LDAC pin to a **low** state, the contents
stored in the DAC’s input registers are transferred
into the DAC’s output registers (VOUT), and both VOUTA
and VOUTB are updated at the same time."

So should you not init the LDAC pin as high and toggle low when you want to latch? I see it is inverted in your library...

RobTillaart, 2023-03-23:
Thanks for this issue.
Sounds like a serious bug so I will look into it asap.

RobTillaart, 2023-03-23:
Created  a develop branch to solve the invert problem (+ fix for setPercentage() + refactor).
Can you give it a try?

nelisbrink, 2023-03-24:
I can confirm, the latch trigger is working correctly now :+1:

RobTillaart, 2023-03-24:
@nelisbrink 
Thanks for testing, I will merge the PR !

closed 2023-03-24 by RobTillaart