RobTillaart/MCP_DAC #32: MCP4922 DAC B control?

jwmaseberg, 2024-09-13:
Hi,
Thanks for this library. I am using your library and MCP4921_standalone.ino code successfully on a nano esp32, and it even controls channel/dac A on a MCP4922 chip. How can I control the B channel on the MCP4922 chip though? 

On the nano/uno I have used code such as https://gist.github.com/equaliser/4358841 for that, if that helps in any way. 

Thanks.

RobTillaart, 2024-09-13:
Thanks for your question.

> How can I control the B channel on the MCP4922 chip though?

The example code cannot do that (yet), a quick look ... think it should be something like

```cpp
voidmcp4922(uint16_t value, uint8_t channel)  //  channel = 0, 1
{
  uint16_t data = 0x3000 | value;
  if (channel == 1) data |= 0x8000;
  digitalWrite(MCP4921_CS_PIN, LOW);
  SPI.beginTransaction(SPISettings(16000000, MSBFIRST, SPI_MODE0));
  SPI.transfer((uint8_t)(data >> 8));
  SPI.transfer((uint8_t)(data & 0xFF));
  SPI.endTransaction();
  digitalWrite(MCP4921_CS_PIN, HIGH);
}
```

Please give it a try

jwmaseberg, 2024-09-13:
It's working great. Thanks for the speedy response!

closed 2024-09-13 by jwmaseberg

RobTillaart, 2024-09-14:
@jwmaseberg 
FYI,
Added **MCP4922_standalone.ino** to the master branch. Did not make a new release (yet).