RobTillaart/MCP_DAC #33: Gain setting

memoir1978, 2025-01-31:
There is a bug in the gain setting code.  The code defaults to gain 1 by or-ing the high bits with 0002.  This is correct.  However, when the set gain function is used it rejects 0 or numbers greater than 2.  It should allow 0 which would set the gain to 2 and only allow gain values 0 or 1.  Alternatively map entered values of 1 or 2 to 0002 and 0000 respectively.

RobTillaart, 2025-01-31:
Thanks for reporting this issue,
Labeled it as bug to give it priority, will look into it right away.

RobTillaart, 2025-01-31:
Checking the relevant  code

Resets sets the default gain to 1x 
```cpp

void MCP_DAC::reset()
{
  _gain     = 1;
...
```

THe setGain only accepts gain == 1x and gain == 2x

```cpp
bool MCP_DAC::setGain(uint8_t gain)
{
  if ((0 == gain) || (gain > 2)) return false;
  _gain = gain;
  return true;
}
```

And only if _gain == 1 the GA bit is set 

```cpp
bool MCP_DAC::write(uint16_t value, uint8_t channel)
...
  if (_gain == 1)   data |= 0x2000;
```

Datasheet page 22  (other are similar)

[image]

Datasheet page 18

_4.1.1.1 Programmable Gain Block
The rail-to-rail output amplifier has two configurable gain options: a gain of 1x (<GA> = 1) or a gain of 2x
(<GA> = 0). **The default value for this bit is a gain of 2 (<GA> = 0).** This results in an ideal full-scale
output of 0.000V to 4.096V due to the internal reference (VREF = 2.048V)._

RobTillaart, 2025-01-31:
@memoir1978 

So my first observation is that the default value of **reset()** does not match the datasheet 4.1.1.1 statement.
That could be fixed, but would give default a higher output voltage and that could ruin a lot of projects.
Imagine after an update the device spits out a 2x higher voltage, it could literally be a breaking change. 

Conclusion: Won't change it to prevent breaking stuff. Add a comment about 4.1.1.1 section.

----

For the rest I do not see a real problem. 
- the gain factor 1 results in a "normal" 1x output  0 - 2.048 Volt
- the gain factor 2 results in a 2x larger output  0 - 4.096 Volt

Using a gain factor 0 would be a gain of zero? no output?

(some moments later)

Think I understand the "confusion"

If you interpret the gain parameter as the bit one should poke into it, then it should be 0 or 1.
Then 0 would map on a gain of 2x and 1 on a gain of 1x 

But as the datasheet talks about a gain of 1 and 2 for a factor of 1x and 2x it seems more logical to me to follow the datasheet.

RobTillaart, 2025-01-31:
@memoir1978 

> It should allow 0 which would set the gain to 2 and only allow gain values 0 or 1. 

Why should setGain() allow the values 0 and 1 in your opinion?

RobTillaart, 2025-01-31:
@memoir1978 

For now I have updated (for a next release) the readme.md file about GAIN to make this parameter more explicit.
It is a summary of the above.

----

### Gain

In **reset()** the gain is default set to 1x as that gives the lowest output voltages.
This is the safest option, although datasheet 4.1.1.1 states default = 2x.

- **bool setGain(uint8_t gain = 1)** gain is 1 (default) or 2. 
Values of 0 or larger than 2 return false and won't change the current setting.
- **uint8_t getGain()** returns gain set, default 1.

Note: The analog output cannot go beyond the supply voltage.  
So if Vref is connected to 5V, gain = 2 will not output 10 Volts.

memoir1978, 2025-02-01:
Rob,I also was thinking maybe we should be OK with the bug!  For those who are feeding back to a 3.3V system, a gain of 2 could be unfortunate. However there are also those who might like a wider swing into a comparator say.  Thanks for looking at it,  I appreciate the incredible work.MikeSent from my iPad

RobTillaart, 2025-02-01:
Thanks again for reporting,
Will not release a new version, the notes are added my local repo so in time they will pop up

closed 2025-02-01 by RobTillaart